vvv-school/vvv17 #2: VM Download

BrutusTT, 2017-01-20:
Hi,

I tried to download the VM image via MegaSync. However, it seems that there is a download limit of 5GB for the free account and the VM image is about 10GB. That posses the isssue of having to pay for being able to download the VM image.

Is there another way to obtain the VM image?

Thanks in advance

jibweb, 2017-01-20:
If you wait ~6h, the download will resume. The VM being over 10Go, that will happen twice !

pattacini, 2017-01-20:
Hi @BrutusTT 

Thanks a lot for reporting this. We obviously apologize for this inconvenience and we are currently going to make the VM available through **FTPS**. The VM will be ready very soon, we just need the time to finish uploading.

Please refer to these [**new instructions**][1].

Stay tuned and sorry again!

[1]: https://github.com/vvv-school/vvv17/wiki/Before%20arriving%20at%20VVV#download-our-virtual-machine

pattacini, 2017-01-22:
The **VM is ready** to be downloaded from our FTP server.
Refer to these [**instructions**][1].

[1]: https://github.com/vvv-school/vvv17/wiki/Before%20arriving%20at%20VVV#download-our-virtual-machine

mbrunettini, 2017-01-22:
The virtual machine file is available at the FTP address described [here ](https://github.com/vvv-school/vvv17/wiki/Before%20arriving%20at%20VVV#download-our-virtual-machine) (the very same link as above)
PLEASE NOTE that you must use a FTP client (like [Filezilla](https://www.google.it/url?sa=t&rct=j&q=&esrc=s&source=web&cd=1&cad=rja&uact=8&ved=0ahUKEwij_erawdXRAhVICMAKHYAIDnwQFggaMAA&url=https%3A%2F%2Ffilezilla-project.org%2F&usg=AFQjCNGk29GabeehesFbZH8JF3yxkKf73A&bvm=bv.144224172,d.bGg) ) to download the virtual machine.

pattacini, 2017-01-24:
Hi @BrutusTT 

Is it going any better with getting the VM?

pattacini, 2017-01-24:
@BrutusTT you should also have received the invitation to join the team. Please, check your email.

pattacini, 2017-01-26:
Closing, feel free to reopen if needed

closed 2017-01-26 by pattacini