vvv-school/vvv17 #3: Compilation error on Install event-driven

ipa-bfb, 2017-01-26:
While trying to compile the code i got the following error:

/home/bfb/robot-code/event-driven/src/core_modules/vUndistortCam/src/vUndistortCam.cpp:188:29: error: ‘getDefaultNewCameraMatrix’ is not a member of ‘cv’
         cv::Mat defCamMat = cv::getDefaultNewCameraMatrix(cameraMatrix, s
                             ^
/home/bfb/robot-code/event-driven/src/core_modules/vUndistortCam/src/vUndistortCam.cpp:190:9: error: ‘undistortPoints’ is not a member of ‘cv’
         cv::undistortPoints(allpoints, mappoints, cameraMatrix, distCoeff
         ^

Any suggestion how to fix it?

pattacini, 2017-01-26:
Hey @arrenglover 

Could you follow this please?

arrenglover, 2017-01-26:
Hi,
The event-driven repository requires OpenCV 2.4, which is the version that is installed when running the step: 
$ sudo apt-get install icub-common. 

If you have done this step then using the flag -DOpenCV_DIR=/usr/share/OpenCV should set the event-driven repository to the correct OpenCV version. Maybe you could check that the files in /usr/share/OpenCV are correct?

If you have a custom install of OpenCV maybe installing icub-common doesn't overwrite a newer version of OpenCV. Another student solved this problem by uninstalling opencv and then reinstalling icub-common. Let me know how you go.

pattacini, 2017-01-29:
Closing for now, feel free to open it again if needed.

closed 2017-01-29 by pattacini

ipa-bfb, 2017-01-30:
Solved! Thanks for the help ;)