vvv-school/vvv17 #4: Error in smoke-test

ipa-bfb, 2017-01-26:
Initially I got the message YARP seems okay!" but in the end I got the following error while make the final testing:

[INFO]  (Test Tutorial Cartesian Interface) reports: yarpmanager is setuping the fixture...
ABORTED: server=/testnode alias=:testnodeiCub_SIM0 cmd=iCub_SIM
Can't execute command because No such file or directory
ABORTED: server=/testnode alias=:testnodeiCub_SIM0 cmd=iCub_SIM
Can't execute command because No such file or directory

Any idea how to solve this?

pattacini, 2017-01-26:
Hi @ipa-bfb,

Supposing you [**installed the system yourself**][1], could you provide a snapshot of your environment?

You can do:
```sh
$ env > env.txt
```
And then attach `env.txt` here in this issue.

[1]: https://github.com/vvv-school/vvv17/wiki/Before%20arriving%20at%20VVV#install-the-code-yourself

ipa-bfb, 2017-01-26:
Yes I installed it by myself. Here it is the file.

Thanks in advance for the help!
[env.txt](https://github.com/vvv-school/vvv17/files/732225/env.txt)

pattacini, 2017-01-26:
The env seems ok. Could you do this test:

On terminal 1:
```sh
$ yarpserver
```

On terminal 2:
```sh
iCub_SIM
```

Does the simulator start off correctly?

jorisguerin, 2017-01-26:
Hi, I am also facing a problem running the smoke-test.
The scene is displaying but the iCub does not move and after a while it crashes with the following error messages : 

[ERROR] (Test Tutorial Cartesian Interface) asserts error on (ret) with exception: yarpmanager (run) cannot setup the fixture because �
[INFO]  (Test Tutorial Cartesian Interface) reports: yarpmanager is tearing down the fixture...
Test suit Test Tutorial Cartesian Interface failed!
Ending test runner.

---------- results -----------
Total number of test suites  : 1
Number of passed test suites : 0
Number of failed test suites : 1
Total number of test cases   : 0
Number of passed test cases  : 0
Number of failed test cases  : 0
./helpers/scripts/smoke-test-cpp.sh : ligne 101 :  3021 Processus arrêté      yarprun --server /testnode
Scanning dependencies of target uninstall
-- Uninstalling "/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/bin/tutorial_cartesian-interface"
-- Uninstalling "/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/ICUBcontrib/applications/tutorial_cartesian-interface-app.xml"
-- Uninstalling "/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/ICUBcontrib/applications/tutorial_cartesian-interface-system.xml"
Built target uninstall
Test suit Test Tutorial Cartesian Interface started...
[INFO]  (Test Tutorial Cartesian Interface) reports: yarpmanager is setuping the fixture...
[ERROR] (Test Tutorial Cartesian Interface) asserts error on (ret) with exception: yarpmanager (run) cannot setup the fixture because �
[INFO]  (Test Tutorial Cartesian Interface) reports: yarpmanager is tearing down the fixture...
Test suit Test Tutorial Cartesian Interface failed!

---------- summary -----------
Total number of test suites  : 1
Number of passed test suites : 0
Number of failed test suites : 1
Total number of test cases   : 0
Number of passed test cases  : 0
Number of failed test cases  : 0
xxxxx Test FAILED xxxxx

If you have an idea how I could fix this problem ...

Thanks!

pattacini, 2017-01-27:
Hi @jorisguerin 
Could you provide a log containing your environment variables?

In the meantime, if you haven't done it yet, please download the VM, at least as fallback solution.

pattacini, 2017-01-27:
Hi @ipa-bfb and @jorisguerin 

Could you also report on the output of these commands:

1. `lsb_release -a`
1. `which iKinCartesianSolver`
1. `which iKinGazeCtrl`
1. `yarp resource --context simConfig --from iCub_parts_activation.ini`

Finally, if you could also dump here the content of the file whose absolute path is found by the last command.

Thanks!

pattacini, 2017-01-27:
cc @pi-q

jorisguerin, 2017-01-27:
Hi @pattacini ,

### Here are the outputs of the commands you provided:

jorisguerin@joris-ubuntu:/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-code$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.1 LTS
Release:	16.04
Codename:	xenial
jorisguerin@joris-ubuntu:/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-code$ which iKinCartesianSolver
/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/bin/iKinCartesianSolver
jorisguerin@joris-ubuntu:/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-code$ which iKinGazeCtrl
/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/bin/iKinGazeCtrl
jorisguerin@joris-ubuntu:/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-code$ yarp resource --context simConfig --from iCub_parts_activation.ini
||| configuring
||| added context simConfig
||| default config file specified as iCub_parts_activation.ini
||| checking [/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-code/iCub_parts_activation.ini] (pwd)
||| checking [/home/jorisguerin/.config/yarp/robots/default] (robot YARP_CONFIG_HOME)
||| checking [/home/jorisguerin/.local/share/yarp/robots/default] (robot YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/robots/default] (robot YARP_CONFIG_DIRS)
||| checking [/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/yarp/robots/default] (robot YARP_DATA_DIRS)
||| checking [/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/iCub/robots/default] (robot YARP_DATA_DIRS)
||| checking [/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/ICUBcontrib/robots/default] (robot YARP_DATA_DIRS)
||| checking [/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-code/codyco-superbuild/build/install/share/codyco/robots/default] (robot YARP_DATA_DIRS)
||| checking [/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/yarp/config/path.d] (robot path.d YARP_DATA_DIRS)
||| found /media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/yarp/config/path.d
||| checking [/home/jorisguerin/.config/yarp/contexts/simConfig] (context YARP_CONFIG_HOME)
||| checking [/home/jorisguerin/.local/share/yarp/contexts/simConfig] (context YARP_DATA_HOME)
||| checking [/etc/xdg/xdg-ubuntu/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [/usr/share/upstart/xdg/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [/etc/xdg/yarp/contexts/simConfig] (context YARP_CONFIG_DIRS)
||| checking [/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/yarp/contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/iCub/contexts/simConfig] (context YARP_DATA_DIRS)
||| found /media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/iCub/contexts/simConfig
||| checking [/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/ICUBcontrib/contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-code/codyco-superbuild/build/install/share/codyco/contexts/simConfig] (context YARP_DATA_DIRS)
||| checking [/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/iCub/contexts/simConfig/iCub_parts_activation.ini] (context)
||| found /media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/iCub/contexts/simConfig/iCub_parts_activation.ini
||| finding file [from]
"/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/iCub/contexts/simConfig/iCub_parts_activation.ini"

### And here is the content of the file:
#### "/media/jorisguerin/82488CC3488CB805/Doctorat/Formations/VVV17/robot-install/share/iCub/contexts/simConfig/iCub_parts_activation.ini"

`/// initialization file for the icub simulation parts
[SETUP]
elevation off
startHomePos on

[PARTS]
legs on
torso on
left_arm on
left_hand on
right_arm on
right_hand on
head on
fixed_hip on

[COLLISIONS]
self_collisions off
covers_collisions off

[SENSORS]
pressure off
whole_body_skin_emul off

[VISION]
cam on

[RENDER]
objects off
screen off

head_cover on
legs_covers on
torso_covers on
left_arm_covers on
right_arm_covers on`

pattacini, 2017-01-27:
Hi @jorisguerin 

It seems all good, therefore we can check it f2f on Sunday afternoon, I'd say.
At any rate, get ready with the VM 😉

pattacini, 2017-01-29:
So, it turned out that `cartesiancontrollerserver` was not enabled.
Once enabled, the test goes smoothly.

closed 2017-01-29 by pattacini