vvv-school/vvv17 #5: issues with opening the VM

kompky, 2017-01-26:
Hi @vvv-school/vvv17-teachers,

I am trying to open the WM but i get the following errors!

[image]

[image]

Best wishes,
Kyveli.

PS. I have windows 8.1 and x64.

mbrunettini, 2017-01-26:
Hi,
you should follow the (1) on the image above: open your bios, go to the security settings and enable "Intel VT-x", then save and reboot.
Let us know if it works for you.

kompky, 2017-01-26:
Thank you very much! Problem solved!

closed 2017-01-26 by kompky

pattacini, 2017-01-26:
Hi @kompky 

Glad that the fix was swift. You should have received the invitation to join the team.
Please check your email.