vvv-school/vvv17 #7: Yarp compilation error

EgorLakomkin, 2017-01-28:
Hello,

While building yarp on Ubuntu 16.04.1 LTS I got this error. I have libtiff5-dev, libtiff5 libraries installed. What could be a problem?

`[ 71%] Linking CXX executable ../../bin/yarpdatadumper
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFReadDirectory@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFWriteEncodedStrip@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFIsTiled@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFOpen@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFReadEncodedStrip@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFSetField@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFWriteScanline@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFGetField@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFScanlineSize@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFNumberOfStrips@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFSetWarningHandler@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFSetErrorHandler@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFReadEncodedTile@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFReadRGBATile@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFClose@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFRGBAImageOK@LIBTIFF_4.0'
/opt/ros/kinetic/lib/libopencv_imgcodecs3.so.3.1.0: undefined reference to `TIFFReadRGBAStrip@LIBTIFF_4.0'
collect2: error: ld returned 1 exit status
src/yarpdatadumper/CMakeFiles/yarpdatadumper.dir/build.make:140: recipe for target 'bin/yarpdatadumper' failed`

towardthesea, 2017-01-28:
Hi @EgorLakomkin ,
It seems that `yarp` found the `opencv` from `ros-kinetic` rather than the normal one. 

Have you tried to install `icub-common` by `$ sudo apt-get install icub-common`, then changed the `OpenCV` path in yarp configuration to the "new" OpenCV from `icub-common`. 

I don't know clearly but there seems some difference between official OpenCV and the one from `ros-kinetic`

Cheers,
Phuong

pattacini, 2017-01-28:
It seems that we are here in a different condition than #3, since on the **VM** Yarp compiles just fine against `/opt/ros/kinect/share/OpenCV-3.1.0-dev` (indeed Yarp is already `opencv-3.x.y` friendly).

However, @EgorLakomkin you could follow @towardthesea's suggestion and give a try to:
```sh
$ cd $ROBOT_CODE/yarp/build
$ rm * -rf
$ cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$ROBOT_INSTALL -DCREATE_GUIS=ON -DCREATE_LIB_MATH=ON -DOpenCV_DIR=/usr/share/OpenCV ../
$ make install
```
which will basically tell Yarp to link against official version of OpenCV for Xenial.

pattacini, 2017-01-28:
@EgorLakomkin we could troubleshoot this problem together tomorrow, but I also suggest you to get ready with the **VM**.

EgorLakomkin, 2017-01-28:
@pattacini @towardthesea Thank you for your help! I advanced further, but still have problems with open cv.

Now it cannot find  headers of OpenCV 
`/opt/ros/kinetic/include/opencv:` No such file or directory

Full error listing:
`[ 94%] Building CXX object src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/src/aboutdlg.cpp.o
cc1plus: warning: /opt/ros/kinetic/include/opencv: No such file or directory [-Wmissing-include-dirs]
[ 94%] Building CXX object src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/src/genericinfodlg.cpp.o
cc1plus: warning: /opt/ros/kinetic/include/opencv: No such file or directory [-Wmissing-include-dirs]
[ 95%] Building CXX object src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/src/loadingwidget.cpp.o
cc1plus: warning: /opt/ros/kinetic/include/opencv: No such file or directory [-Wmissing-include-dirs]
[ 95%] Building CXX object src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/src/main.cpp.o
cc1plus: warning: /opt/ros/kinetic/include/opencv: No such file or directory [-Wmissing-include-dirs]
In file included from /home/egor/robot-code/yarp/src/yarpdataplayer/include/utils.h:32:0,
                 from /home/egor/robot-code/yarp/src/yarpdataplayer/include/mainwindow.h:16,
                 from /home/egor/robot-code/yarp/src/yarpdataplayer/src/main.cpp:1:
/home/egor/robot-code/yarp/src/yarpdataplayer/include/worker.h:33:18: fatal error: cv.h: No such file or directory
compilation terminated.
src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/build.make:169: recipe for target 'src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/src/main.cpp.o' failed
make[2]: *** [src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/src/main.cpp.o] Error 1
CMakeFiles/Makefile2:3419: recipe for target 'src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/all' failed
make[1]: *** [src/yarpdataplayer/CMakeFiles/yarpdataplayer.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2`

pattacini, 2017-01-28:
Hi @EgorLakomkin 

Did you make sure you purged the previous compilation products outright and started from scratch again?

I'm asking 'cause if you configure Yarp to compile against `/usr/share/OpenCV` as suggested above (i.e. the cmake option `-DOpenCV_DIR=/usr/share/OpenCV`), then you should no longer have any reference to `/opt/ros/kinetic/include/opencv`.

towardthesea, 2017-01-28:
@EgorLakomkin please remove the `CMakeCache` or the whole old `build` folder before following the manual of @patacini

EgorLakomkin, 2017-01-28:
It worked, thank you!

closed 2017-01-28 by EgorLakomkin

pattacini, 2017-01-28:
Great!