vvv-school/vvv17 #8: Join the team and send the presentation

pattacini, 2017-01-28:
@AlexisBillier and @damianomal
Please, accept the invitation to join the team that you should have received in your mail box.

@alexarossi @anatanevska @bmagyar @ipa-bfb @EgorLakomkin @BrutusTT @kompky @LuSeKa @pi-q 
Please, send us the one-page pdf presentation.

pi-q, 2017-01-28:
Hi Ugo,

Here is my presentation. Let me know if you need anything more.

Cheers,

Marie

pattacini, 2017-01-28:
Thanks @pi-q, just uploaded!

alexarossi, 2017-01-28:
Ciao Ugo,

Attached you can find my presentation. Sorry for the delay.

Cheers,
Alessandra

bmagyar, 2017-01-28:
Hiya, 
Mine is on #9 
Cheers,
Bence

pattacini, 2017-01-29:
Ciao Alessandra,

Didn't receive your slide yet. Do you mind drop me an ✉️ at this [**address**][1] with the slide attached?

See you.

[1]: mailto:vvv-admin@iit.it

pattacini, 2017-01-29:
Reminder

- [ ] @AlexisBillier to join the team
- [ ] @damianomal to join the team
- [ ] @ipa-bfb to send the presentation
- [ ] @BrutusTT to send the presentation
- [ ] @kompky to sent the presentation

pattacini, 2017-01-30:
We're all done, thanks!

closed 2017-01-30 by pattacini