vvv-school/vvv17 #10: VM and MATLab location

BrutusTT, 2017-01-29:
Hi,

I am using the VM and had to install the Matlab in a directory in my home. How can I tell cmake to find it there? The missing variables its complaining about seem to be not accessible by providing them as environment variables or cmake switches.

How would I set Matlab_ROOT_DIR for example.

pattacini, 2017-01-29:
@francesco-romano would you be willing to follow this?

traversaro, 2017-01-29:
Hi @BrutusTT , in our projects we are using the `FindMatlab.cmake` available in the latest versions of CMake, imported locally to make sure that is available also for users with old versions of CMake.

If Matlab is installed in a non-canonical directory, it should be sufficient to just set the `Matlab_ROOT_DIR` CMake option, as explained in CMake documentation : https://cmake.org/cmake/help/v3.7/module/FindMatlab.html#variable:Matlab_ROOT_DIR . If this is not working for you, can you provide the exact error message that you are receiving during configuration? Thanks!

BrutusTT, 2017-01-29:
Thanks for the help. 

I managed to get it working by setting adding the bin directory for matlab to the PATH variable. For some reason the Matlab_ROOT_DIR was not picked up by CMake. Any way it works now.

closed 2017-01-29 by BrutusTT