vvv-school/vvv17 #14: codyco-superbuild building error. Expected and actual hash mismatch for YCM_BOOTSTRAP HASH mismatch

EgorLakomkin, 2017-01-29:
Hello,

I faced an error while building codyco-superbuild on Ubuntu 16.04.1 LTS. I have attached the error listing below.

___________________________

-- YCM not found. Bootstrapping it.
-- Downloading YCMEPHelper.cmake - Attempt 1 of 3
-- Downloading YCMEPHelper.cmake - SUCCESS
CMake Warning (dev) at build/CMakeFiles/YCMEPHelper.cmake:164 (message):
  YCM_BOOTSTRAP HASH mismatch

    for file: [/home/egor/robot-code/codyco-superbuild/cmake/IncludeUrl.cmake]
      expected hash: [20835806b1ccb6950c8d235f86c16c938064a4d4]
        actual hash: [5faefc77acbbe7314d1ddd4560dd48b0eef291c5]

  Perhaps it is outdated or you have local modification.  Please consider
  upgrading it, or contributing your changes to YCM.

Call Stack (most recent call first):
  build/CMakeFiles/YCMEPHelper.cmake:1342 (_ycm_hash_check)
  cmake/YCMBootstrap.cmake:106 (ycm_bootstrap)
  CMakeLists.txt:92 (include)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at build/CMakeFiles/YCMEPHelper.cmake:164 (message):
  YCM_BOOTSTRAP HASH mismatch

    for file: [/home/egor/robot-code/codyco-superbuild/cmake/YCMBootstrap.cmake]
      expected hash: [8cad902fcd79fb2dc70d14af73d74fdd8c73df94]
        actual hash: [c696c89613248c243076b4d8a3dcc3bdd3e2e590]`

@francesco-romano @traversaro

traversaro, 2017-01-29:
I @EgorLakomkin, 
 this is actually just a warning, i.e. you can continue we the compilation of the superbuild perfectly fine. 

It is due to https://github.com/robotology/ycm/issues/93 and https://github.com/robotology/codyco-superbuild/issues/162 , and it is actually quite confusing to the end users, as you are not the first one that reported it as an error. To avoid further confusion, I disabled the warning with https://github.com/robotology/codyco-superbuild/commit/6bbf5349d0417c67c2c2b2b07da57c5941c3bb1e until we actually fix the issue, if you update the `codyco-superbuild` repository (i.e. `git pull` in the superbuild directory) the warning should disappear.

EgorLakomkin, 2017-01-30:
@traversaro Thank you! It worked, did not realize that it was just a warning.

closed 2017-01-30 by EgorLakomkin