vvv-school/vvv17 #18: Register to book time slots on the robots

pattacini, 2017-01-30:
Dear @vvv-school/vvv17-students 

You should have received another email asking you to join our **robot booking system**.
Please proceed by following the instructions.

Once done, you'll be able to book time slots on the iCub for testing your assignments for real 😉 

You can reach the **booking dashboard** from within our [Wiki](https://github.com/vvv-school/vvv17/wiki).

ipa-bfb, 2017-01-30:
How should we organize the time? Are we going to create teams?

pattacini, 2017-01-30:
I can give an answer for my course on kinematics (3rd day), for which I tend to follow this approach:

There'll be two assignments in the afternoon, whose the second can be run on the robot too.

The assignments is on an individual basis, so I'm going to check first whether the student will get the grade on the simulator and then I'll give him green light to book the robot.

Therefore, no teams for me, but other teachers may decide differently.

pattacini, 2017-01-31:
For whoever hasn't done this yet: please, do it before tomorrow.

cc @vvv-school/vvv17-students

closed 2017-02-02 by pattacini