vvv-school/vvv17 #22: Assignment on Yarp producer-consumer

apaikan, 2017-01-30:
Click and edit and score 5: 
https://classroom.github.com/assignment-invitations/9ff783408efb2a4c3ef7ee53277977a1

pattacini, 2017-01-30:
🎖 to @EgorLakomkin for being the first to complete the assignment 😄

AlexAntn, 2017-01-30:
Hello @apaikan ,

We detected that the test script doesn't detect when we don't properly clear the Bottle from the BufferedPort<Bottle> (in the producer side), which might lead to the Bottle growing forever!

apaikan, 2017-01-30:
@AlexAntn  well seen! Today unfortunately we could not manage to completely go through the tutorial of yarp::os::BufferedPort to cover this part. So i kept is simple by copying the output into the input!  We will cover it tomorrow during the assignment. Btw, +1 for the test! 👍

closed 2017-01-31 by pattacini