vvv-school/vvv17 #28: Sharing folder between windows and our VM machine using virtualBox

vtikha, 2017-01-31:
Just a small note to all, 

if you need to share a folder between a host (Windows/Mac) and the distributed Virtual Machine using Virtual box, please follow these simple steps:
```
 - Devices -> Shared Folders -> Shared Folder Settings
 - Click on the Add Folder icon. 
 - Select the Host Folder and give it a Folder Name, 
 - Tick the "Auto Mount" check-box.
```

Now the magical command to will fix all permission errors:
```
sudo adduser icub vboxsf
```
Please **log out** and log back in for it to take effect.

pattacini, 2017-01-31:
cc @vvv-school/vvv17-students

vtikha, 2017-01-31:
@pattacini this has nothing to do with ros 😄 just VM machine folder sharing

pattacini, 2017-01-31:
I just tried 😊

closed 2017-01-31 by pattacini