vvv-school/vvv17 #32: [Dynamics course] Update to VM and instructions

francesco-romano, 2017-01-31:
Dear @vvv-school/vvv17-students 

some more actions are needed to run the tutorials and assignments (especially the smoke-tests on your local machine) of the dynamics course.

---
The following environment variables were not "complete" in the VM and instructions, so please update as in the following.
Substitute the following lines in your `.bashrc_dev` file:
```sh
export GAZEBO_PLUGIN_PATH=${ROBOT_CODE}/codyco-superbuild/build/install/lib
export GAZEBO_MODEL_PATH=${ROBOT_CODE}/codyco-superbuild/build/install/share/gazebo/models
```

with 
```sh
source /usr/share/gazebo/setup.sh
export GAZEBO_PLUGIN_PATH=${ROBOT_CODE}/codyco-superbuild/build/install/lib
export GAZEBO_MODEL_PATH=${ROBOT_CODE}/codyco-superbuild/build/install/share/gazebo/models
export GAZEBO_RESOURCE_PATH=${GAZEBO_RESOURCE_PATH}:${ROBOT_CODE}/codyco-superbuild/build/install/share/gazebo/worlds
```
(See [this commit](https://github.com/vvv-school/vvv-school.github.io/commit/5591d22e190bd5437e504fc665bd21fefe40f14a))

Also substitute the following line
```sh
export MATLABPATH=${ROBOT_CODE}/codyco-superbuild/build/install/mex:${ROBOT_CODE}/codyco-superbuild/build/install/share/WB-Toolbox:${ROBOT_CODE}/codyco-superbuild/build/install/share/WB-Toolbox/images
```
with

```sh
export MATLABPATH=${ROBOT_CODE}/codyco-superbuild/build/install/mex:${ROBOT_CODE}/codyco-superbuild/build/install/share/WB-Toolbox:${ROBOT_CODE}/codyco-superbuild/build/install/share/WB-Toolbox/images:${ROBOT_INSTALL}/simmodels
```
(See [this commit](https://github.com/vvv-school/vvv-school.github.io/commit/fe59ededf6759b2b9f41406e614e1f095c3eaca6))

Unfortunately, Matlab has some issues with libraries, i.e. it embeds some system libraries and this can conflict with other libraries.
You may need to run the following commands (thus hiding the MATLAB library)
```sh
sudo mv /usr/local/MATLAB/R2016b/sys/os/glnxa64/libstdc++.so.6.0.20 /usr/local/MATLAB/R2016b/sys/os/glnxa64/libstdc++.so.6.0.20.bak
```

---
Finally some updates have been done to the gazebo models so you may have to update the `codyco-superbuild`: 
```sh
$ cd ${ROBOT_CODE}/codyco-superbuild/build
$ make update-all
$ make
```

Hopefully this is enough to run gazebo and the demos.

Arya07, 2017-02-01:
never leave your laptop unlocked!

gmmarconi, 2017-02-01:
Note: the file to edit is ```.bashrc-dev``` inside the home folder.

closed 2017-02-02 by pattacini