vvv-school/vvv17 #43: Assignment on yarpWholeBodyInterface

francesco-romano, 2017-02-02:
@vvv-school/vvv17-students 
here is the first assignment with the C++ library (yarpWholeBodyInterface)

https://classroom.github.com/assignment-invitations/562e2324da65e834ec7709b91e780cfd

francesco-romano, 2017-02-02:
Error on the README of the assignment.

Call the reference port as 
`/computed-torque/qDes:i`

francesco-romano, 2017-02-02:
@vvv-school/vvv17-students 
This applies also to the Simulink assignment: https://github.com/vvv-school/vvv17/issues/44

You have to stabilize a set point (i.e. desired velocity is zero).
This reference come though a port which in the c++ assignment must be called `/computed-torque/qDes:i`, while the simulink part already has that block (please do not change the simulink model name)

pattacini, 2017-02-02:
Honor to @fabian-paus for being the first in scoring 5 point with this assignment 🏅

closed 2017-02-02 by pattacini