vvv-school/vvv17 #46: Smoke test the code without cloning helper tools

pattacini, 2017-02-03:
Dear @vvv-school/vvv17-students 

@bmagyar gave us a good suggestion on how to avoid hitting the network altogether while smoke testing locally your code.

To this aim, within the assignment/tutorial smoke-test directory do:
1. Run once `test.sh`
1. `cp test.sh test-noclone.sh`
1. Edit `test-noclone.sh` by:
    1. Comment out line **16** (e.g. [this one][1])
    1. Comment out line **21** (e.g. [this one][2])
1. From now on, launch `test-noclone.sh` in place of the original `test.sh`

[1]: https://github.com/vvv-school/assignment_make-it-roll/blob/master/smoke-test/test.sh#L16
[2]: https://github.com/vvv-school/assignment_make-it-roll/blob/master/smoke-test/test.sh#L21

bmagyar, 2017-02-03:
@pattacini We also need to edit line **18** [here](https://github.com/vvv-school/assignment_make-it-roll/blob/master/smoke-test/test.sh#L18) and remove `mkdir build &&`

bmagyar, 2017-02-03:
Once I've done that tests are pretty swift, thanks for the update!

pattacini, 2017-02-03:
@bmagyar you're right, thanks for the amendment; going to put in the post above. 👍

pattacini, 2017-02-05:
The original upstream repositories have been all updated with a new smoke test script that first verifies whether we have already a local version of the helper tools: if yes it uses it, otherwise it starts cloning.

@vvv-school/vvv17-teachers I should have covered all the cases (i.e. tutorials, assignments, solutions), but please double check.

closed 2017-02-05 by pattacini

pattacini, 2017-02-05:
@vvv-school/vvv17-students you may want to update your local copy of the test with this [**newest version**][1].

[1]: https://github.com/vvv-school/tutorial_cartesian-interface/blob/master/smoke-test/test.sh