vvv-school/vvv17 #56: Smoke test for assignment

vtikha, 2017-02-03:
@vvv-school/vvv17-students 
please be aware that the smoke test has changed. To carry out a proper test you please apply this [patch][1] to your local copy of the test.cpp

[1]: https://github.com/vvv-school/assignment_closest-blob/commit/7565d5401fbac48fa50ce92bb5dce26ec12b292f

vtikha, 2017-02-03:
@vvv-school/vvv17-students 
to simplify things copy and paste the entire test.cpp you can find [here](https://github.com/vvv-school/assignment_closest-blob/blob/master/smoke-test/test.cpp)

BarisSerhan, 2017-02-03:
:)
[image]

pattacini, 2017-02-04:
So, what was the final record of yesterday course? 😉

closed 2017-02-04 by pattacini

BarisSerhan, 2017-02-04:
it was 0.9666, not so bad i think

ipa-bfb, 2017-02-04:
I got 0.971 ;)