vvv-school/vvv17 #59: Script for drawing team leaders for the final assignment

pattacini, 2017-02-04:
So, it looks like we'll have **lots of candidates as leaders** of the teams we'll form to tackle the final assignment, since many of you @vvv-school/vvv17-teachers are performing really well in this school, and the rest are really running up closely (check out the [**ranking**][1]) 🎉 

Therefore we'll have to pick up those at the top of the final gradebook and possibly also **draw some others**. We could do that at the old manner with names written on small pieces of paper or, even better, with a **script**! 😉 

**Is anybody willing to write that script ?** That would be great! 💻

Requirements:
- We'll run the script on **Ubuntu >= trusty**.
- You can use **bash**, **python**, **ruby** ...
- You'll have to **`wget` final-gradebook.md** and process it:
    - `wget -O final-gradebook.md https://raw.githubusercontent.com/vvv-school/vvv17/master/final-gradebook.md`
- We would need the script ready and functioning by **Monday morning**.

[1]: https://vvv-school.github.io/vvv17/final-gradebook.html

pattacini, 2017-02-04:
Let us know about your commitment to this task.

damianomal, 2017-02-06:
Hi Ugo, 

here you can find a script which should do the job: 
https://github.com/damianomal/vvv17-get-some-leaders

Hope it helps!

See you tomorrow, 
Damiano

pattacini, 2017-02-06:
Fantastic @damianomal 
I gave just a try to it and it seems it works nicely in different conditions of the gradebook. In particular, it's not always drawing your username 😉 

Why don't you open a PR and locate this script in `./scripts` directory of this repository with default file `../final-gradebook.md`?

closed 2017-02-06 by pattacini