vvv-school/vvv17 #62: Cannot push to wiki as student

LuSeKa, 2017-02-04:
Tried to push to the wiki (change in Fun Zone), but push access was denied. Either something is wrong with the permissions or I am doing something wrong...

LuSeKa, 2017-02-04:
@pattacini

pattacini, 2017-02-04:
What do you mean with **push**? Thorugh git?
You should just **edit** it. Don't you see the Edit button on the top right of the page?

LuSeKa, 2017-02-04:
No, no edit button there :(

pattacini, 2017-02-04:
Damn'
According to this [**post**][1], students should have permission to edit wiki...
I might have misunderstood that. Let me better inquire.

[1]: https://help.github.com/articles/repository-permission-levels-for-an-organization

pattacini, 2017-02-04:
@LuSeKa done! Now you can edit the wiki.

pattacini, 2017-02-04:
This post helps: https://help.github.com/articles/changing-access-permissions-for-wikis

closed 2017-02-04 by pattacini