vvv-school/vvv17 #65: Social Dinner!

ChiaraBartolozzi, 2017-02-04:
The Social Dinner will be at [**Ristorante La Darsena**](https://www.google.it/maps/place/La+Darsena/@44.3358157,9.2141819,17z/data=!3m1!4b1!4m5!3m4!1s0x12d4a0230c3607ef:0xfd291eb596ca6d33!8m2!3d44.3358157!4d9.2163706) on Wed. 8th @ 8pm.

Please check your name if you plan to attend (no dinner will be offered at the hotel) and add a note for special dietary needs.

- [ ] Jean-Baptiste Weibel
- [ ] Nolwenn Briquet-Kerestedjian
- [ ] Joris Guérin
- [ ] Bruno Brito
- [ ] Ingo Keller 
- [ ] Bence Magyar 
- [ ] Massimiliano Iacono
- [ ] Elisa Maiettini 
- [ ] Gian Maria	Marconi (no fish)
- [ ] Ana Tenevska
- [ ] Adrien Malaisé (vegetarian)
- [ ] TAI PANG, Lawrence Chen
- [ ] Nuno Duarte 
- [ ] Pedro Vicente
- [ ] Lukas Kaul  (Vegetarian)
- [ ] Fabian Paus
- [ ] David Busson 
- [ ] Serhan Barış
- [ ] Marie Charbonneau (soja, gluten, mushroom, corn)
- [ ] Dong Hai Phuong Nguyen
- [ ] Alexis Billier 
- [ ] Svenja Stark 
- [ ] Damiano Malafronte 
- [ ] Chandrakant Bothe
- [ ] Egor Lakomkin 
- [ ] Mohammad Ali Zamani 
- [ ] Alessandra Rossi
- [ ] Alexandre Antunes 
- [ ] Kyveli Kompatsiari 
- [x] Chiara Bartolozzi
- [ ] Ugo Pattacini
- [ ] Vadim Tikhanoff
- [ ] Francesco Nori
- [ ] Lorenzo Natale
- [ ] Francesco Romano
- [ ] Daniele Pucci
- [ ] Silvio Traversaro
- [ ] Arren Glover
- [ ] Ali Paikan
- [ ] Giulia Pasquale
- [ ] Raffaello Camoriano

ChiaraBartolozzi, 2017-02-04:
cc @vvv-school/vvv17-students @vvv-school/vvv17-teachers

pattacini, 2017-02-04:
Dear @vvv-school/vvv17-students 
It turned out that you don't have the permission to edit the post, hence we've just made available the same exact table within the wiki.

please edit it here: https://github.com/vvv-school/vvv17/wiki/Social%20Dinner

ChiaraBartolozzi, 2017-02-07:
Dear @vvv-school/vvv17-students and @vvv-school/vvv17-teachers 

Please remember to update this [page](https://github.com/vvv-school/vvv17/wiki/Social%20dinner) before lunch if you wish to join our final dinner.

closed 2017-02-09 by pattacini