vvv-school/vvv17 #68: Download the iCW dataset for the Deep Learning course

GiuliaP, 2017-02-05:
Hi @vvv-school/vvv17-students @vvv-school/vvv17-teachers ,

in order to get ready for the Deep Learning course on tuesday, you should start downloading a dataset of images that will be used in the hands on session. Since it is around 500MB, and you all need to download it, it may be better if you start it in advance.

You can follow the instructions at this [link](https://github.com/vvv-school/tutorial_dl-tuning), which include also a simple test, that you may want to perform after downloading the dataset, in order to be sure that your system is setup correctly to run `caffe` and fine-tune deep CNNs.

Thank you!
See you on tuesday :)

Giulia

vtikha, 2017-02-05:
If, for any reason, you run into problems with the download, we will also provide the dataset in a usb key on monday morning

pattacini, 2017-02-06:
cc @vvv-school/vvv17-students 

A gentle reminder to download the new dataset for deep learning 😉

closed 2017-02-07 by pattacini