vvv-school/vvv17 #69: [Assignment orientation-Filter]

vicentepedro, 2017-02-05:
There is an issue with the initialization of the parameters inside the module.

check l.29 of the cpp file (and the lines below)
`return spikingmodel.initialise(...)`

And make sure the order of the parameters is the same as in:
l.46
`bool spikingModel::initialise(...)`

They are not in the correct order by default. the theta value should be the last parameter.

If you want to add some info to this issue @arrenglover.

vicentepedro, 2017-02-05:
@vvv-school/vvv17-students 
Maybe it's a good idea to mention the students =P

towardthesea, 2017-02-05:
FYI
https://github.com/vvv-school/assignment_event-orientation-filter/blob/master/src/event-orientation-filter.cpp#L33

pattacini, 2017-02-06:
Supposedly, this has been fixed.
Thanks!

closed 2017-02-06 by pattacini