vvv-school/vvv17 #70: Team contest

pattacini, 2017-02-05:
I'm not in Santa yet, but I guess you're all working on the assignment now, right? 😉 

We've got a news for you. As anticipated a couple of days ago, there'll be a team contest running the last days and tomorrow we're going to create teams, then.

We've just provided some information on the event. [**Check it out**][1] 🎉 

More news to be announced. Stay tuned.

cc @vvv-school/vvv17-students 

[1]: ../wiki/Team-contest

vtikha, 2017-02-06:
if anyone figured out what @pattacini meant by `being Santa` please let me know 😄

pattacini, 2017-02-06:
🎅

pattacini, 2017-02-06:
Hey @vvv-school/vvv17-students 

The gradebooks will be closed at 12:45.
Then, we'll choose the team leaders using @damianomal's [**script**](../blob/master/scripts/draw-team-leaders.py).

vtikha, 2017-02-06:
Tick Tock ⏳

pattacini, 2017-02-06:
Teams are now ready!

check it out here: https://github.com/orgs/vvv-school/teams 

@vvv-school/vvv17-teachers @vvv-school/vvv17-students

pattacini, 2017-02-06:
Let the contest begin!

You can work on the final assignment this evening, tomorrow evening and the whole day of Wednesday!

Good luck!

pattacini, 2017-02-06:
Let me remind you that each team has its own private repository to work in.

https://github.com/vvv-school/vvv17/wiki/Team%20contest#repositories

pattacini, 2017-02-06:
cc @vvv-school/vvv17-students

closed 2017-02-06 by pattacini

pattacini, 2017-02-07:
Result of drawing:

1. @Arya07 team leader of @vvv-school/vvv17-team-green 
1. @Iaxama team leader of @vvv-school/vvv17-team-orange 
1. @bmagyar team leader of @vvv-school/vvv17-team-blue