vvv-school/vvv17 #74: Blue team

bmagyar, 2017-02-06:
Hello @vvv-school/vvv17-team-blue !

I would like to ask everyone to quickly read through [this](https://github.com/vvv-school/vvv17/wiki/Team-contest) and (while you are crunching matlab tutorials) just think about potential demos or what you'd be interested in doing aligned with what you are good with. 

First thing we should do is to come together and discuss ideas this evening.

bmagyar, 2017-02-06:
Coming up!

bmagyar, 2017-02-06:
[Shared google doc to dump ideas into](https://docs.google.com/document/d/1aJLsfLf6sjJVVxXDRZr-BTYVdUqlVetG82VZoDWVkF8/edit?usp=sharing)

**Trolls keep out**

closed 2017-02-06 by bmagyar

pattacini, 2017-02-06:
Please use the private repository instead! 😉 

https://github.com/vvv-school/vvv17-demo-team-blue

vicentepedro, 2017-02-06:
@vvv-school/vvv17-team-blue, you should use this repository:
https://github.com/vvv-school/vvv17-demo-team-blue
It's a private repo, so, the other teams will not be able to see anything ;)