vvv-school/vvv17 #82: Greetings from VVV17

pattacini, 2017-02-09:
As @vvv-school/vvv17-teachers we'd like to thank all of you @vvv-school/vvv17-students for having made this school so special with your attitude, interest, involvement and all the effort you put to stay on the ball with the lessons and the assignments.

It's been a real pleasure to organize the school and spend these amazing days together and now I personally feel a bit sad that all is over. It seems you enjoyed this basecamp as much as we did, as we received lots of very good feedback from many of you along with nice suggestions to further improve this event and we are so happy about that. We also hope you've both learned something useful for your careers and earned great moments and experiences to keep in your memories.

We wish you all the best!

Yours,
@vvv-school/vvv17-teachers

DanielePucci, 2017-02-09:
:)

bmagyar, 2017-02-11:
Thank you @vvv17-teachers for this well-executed winter school! It was a
great experience for me and I wish no less for the coming ones ;)

BrutusTT, 2017-02-11:
I also would like to thank for this great winter school.

After seeing how much can be achieved if inspired students are given 48h to come up with a nice iCub demo, some off us already discussed a small addition to the school. 

What about: A weekend in the summer ... Winter School Alumnis ... 48h (without distractions ;) )...  and an iCub Hackathon? It also would give us students a chance to gather again and strenghten our collaboration skills.

pattacini, 2017-02-11:
Hi @bmagyar and @BrutusTT 
It's great to hear you 😄 

The hackathon would be a relly nice follow-up indeed, but I think we should keep an eye to funding too and I don't know if there's a budget to make this idea come true.

Once possibility would be to invite you guys at IIT and reserve our iCub's for some days to the hackathon. Don't know if it's doable, though. Let me ask @lornat75 to chime in.

Of course, people researching at institutions hosting iCub shouldn't wait longer, but should rather plan to continue refining the demos collaborating via GitHub 😉

ipa-bfb, 2017-02-13:
I share the same opinion as @bmagyar and @BrutusTT. It is incredible how much we could learn in less than 2 weeks! And nothing with be possible without you awesome organization and help ;) It was a pleasure to meet you!

Best,
Bruno!

nolbriker, 2017-02-13:
Like the previous guys, I wanted to congratulate all the teachers for this great organization of the school. I can't imagine the incredible amount of preparation work! You were so involved during the school that we tried to do the same by revealing unknown programming capabilities until late into the night :)

I wish you the same success for the next editions, and hope to see you soon at a conference!
Nolwenn

lornat75, 2017-02-13:
Thank you all for the work and efforts, surviving such compressed 10 days of lectures and tutorials was probably all but easy :) 

For the hackaton/workshop, it is a good idea we may consider it in the future. It may be difficult to find financial support though. 

Please take the final effort to fill the questionnaire I have circulated, I am sure you can come up with useful feedback!

closed 2017-09-15 by pattacini