vvv-school/vvv17 #84: iCub_SIM component problem

accateo, 2017-02-19:
Hi,
I don't know if this is the right place to ask this question. 
I'm following your YARP tutorial and I'm at the point that I have to run iCub_SIM. I run it, and all it's ok. When I run "yarpmotorgui", it appears a popup window that asks me what component of the iCub control. I select all the components (i have also tried to select some), I click on OK, but appear different windows that tell me "Error while opening device /icub/head".
The same thing happens with Yarpmanager.
Why this happens?
Thanks

vtikha, 2017-02-19:
Hi @accateo The only thing you need to do here is write icubSim instead of icub in the popup window when you select which component you need.

pattacini, 2017-02-19:
Hi @accateo ,

Next time, please use our [**Q&A**][1] system instead 😉 

[1]: https://github.com/robotology/QA

closed 2017-02-19 by pattacini

accateo, 2017-02-19:
Thanks! It works! Sorry for the wrong place.